va3wam/SegbotSTEP #10: Build a power circuit for the ESP8266 and motors

va3wam, 2017-11-16:
Create a power circuit so that the robot can run without being tethered to a laptop

nerdoug, 2017-11-20:
I think we don't need a 3.3 V voltage regulator. If we connect 5V to either the BAT or the USB pins on the ESP8266, the onboard regulator will produce 3.3 V, the same way it does if a USB console cable is connected.

va3wam, 2017-11-20:
Andrew

va3wam, 2017-11-20:
Oops, hit send on that last note by accident. Great news about the power, hoping we do not cook our boards!

I did not work on the robot when you left, but I did look into the serial over ip stuff a bit. I think I will continue working on the ip socket code angle as that serial wifi stuff seems to need special boot loaders and possibly hardware to make work.

Andrew